arduino/ArduinoCore-API #22: Unit tests?

leg0, 2018-11-16:
I was wondering if there are any plans to start using unit tests in this project? Any plans to run them in a continuous integration server? There are some free CI servers, for example Travis CI, Azure Devops, etc.

As part of a pull request I would like to provide a set of unit tests to demonstrate that my changes actually work (at least the way I intended them to).

facchinm, 2018-11-19:
Hi @leg0 ,
absolutely :slightly_smiling_face: We already have many unit test running on real boards but a community driven, hardware agnostic test suite would be awesome. 
In the meantime, feel free to provide any test sketch with your PRs, maybe based on  https://github.com/mmurdoch/arduinounit .

ianfixes, 2021-01-18:
Is this a request to unit test the library itself, or to provide an ability to mock this library for the purposes of unit testing 3rd-party Arduino libraries?

Asking as the maintainer of [the `arduino_ci` project](https://github.com/Arduino-CI/arduino_ci).

leg0, 2021-01-18:
In short, it's a request to have a place to put unit tests for a code in this repo.

When I created this question, i had a PR that I had some tests, but I did not see any unit testing framework being used in this project. So including a set of unit tests that nobody would run seemed like a waste, and as a result that test code would probably rot and not be able to run anyway after a while. If there are plans to add unit testing to this project, it would be helpful to know which testing framework is going to be used. And once that is available, running the tests as part of CI pipeline would then become a possiblity.

per1234, 2021-01-18:
Hi @leg0. Some nice advancements have been made in this area since the time you created the issue. You can see the unit tests here:
https://github.com/arduino/ArduinoCore-API/tree/master/test/src

They are using the [Catch2 test framework](https://github.com/catchorg/Catch2).

These test are run by the[ GitHub Actions](https://github.com/features/actions)-based CI system:
https://github.com/arduino/ArduinoCore-API/blob/master/.github/workflows/unit-tests.yml
https://github.com/arduino/ArduinoCore-API/actions?query=workflow%3A%22Unit+Tests%22
and code coverage reports are published on Codecov (96% coverage!):
https://codecov.io/gh/arduino/ArduinoCore-API

closed 2021-05-25 by leg0