arduino/ArduinoCore-API #24: I2S API missing

ChrisMicro, 2018-12-11:
It is more and more common on 32Bit MCUs to support audio applications. 
What is really missing is the unification of the I2S-API on the different platforms.

A good start would be this API:
https://danieleff.github.io/STM32GENERIC/i2s/

sandeepmistry, 2018-12-12:
There's an official I2S library for the SAMD core: https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/I2S

Formal documentation can be found here: https://www.arduino.cc/en/Reference/I2S

ChrisMicro, 2018-12-14:
Thank you for the hints.
What do you think of including this API functions here in the repository?

sandeepmistry, 2018-12-14:
@ChrisMicro I think so, @cmaglie @facchinm what do you think about adding a HardwareI2S.h file?

facchinm, 2018-12-14:
Totally agree, we should add it :slightly_smiling_face: