arduino/ArduinoCore-API #25: Breakage caused by PinStatus and PinMode types

per1234, 2019-01-02:
This project changes `LOW`, `HIGH`, `INPUT`, `INPUT_PULLUP`, and `OUTPUT` from macros to enums:
https://github.com/arduino/ArduinoCore-API/blob/e1eb8de126786b7701b211332dda3f09aa400f35/api/Common.h#L10-L23

I'm concerned that this will cause breakage of a significant amount of existing code.

An example is the popular [Keypad library](https://playground.arduino.cc/code/keypad). Compilation of both the [original](https://playground.arduino.cc/uploads/Code/keypad.zip) library and the [version in Library Manager](https://github.com/Chris--A/Keypad) fails once this change is made.
```
In file included from E:\electronics\arduino\libraries\Keypad-master\examples\HelloKeypad\HelloKeypad.ino:10:0:

E:\electronics\arduino\libraries\Keypad-master\src/Keypad.h: In member function 'virtual void Keypad::pin_write(byte, boolean)':

E:\electronics\arduino\libraries\Keypad-master\src/Keypad.h:81:81: error: cannot convert 'boolean {aka bool}' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

  virtual void pin_write(byte pinNum, boolean level) { digitalWrite(pinNum, level); }
```
This commonly used code will also now fail:
```c++
digitalWrite(13, !digitalRead(13));  // toggle pin 13
```
```
toggle:2:36: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

   digitalWrite(13, !digitalRead(13));  // toggle pin 13
```
I understand that the root cause of these errors is bad code and that any code which followed best practices will have no problems with this change. However, I fear there is a lot of bad code in widespread use that currently works fine. In the case of the Keypad library, it is unlikely it can even be fixed since Chris--A has gone AWOL. I'm sure there are other such abandoned projects.

I do like the spirit of this change (though lumping `CHANGE`, `FALLING`, and `RISING` into `PinStatus` is questionable). I'm open to being convinced that it's worth the breakage and, if so, I'm willing to help ease the transition by providing user support and submitting PRs to fix broken code. I just think this warrants some consideration before ArduinoCore-API goes into more widespread use.

Some previous discussion on the topic:
- http://forum.arduino.cc/index.php?topic=455579 (from [here](http://forum.arduino.cc/index.php?topic=455579.msg4001965#msg4001965) onward)
- https://forum.arduino.cc/index.php?topic=584322

PaulStoffregen, 2019-01-03:
I can confirm more code in the wild which does an ifdef check on INPUT_PULLUP (and falls back to pinMode & digitalWrite), because INPUT_PULLUP was added to Arduino later than the others.

matthijskooijman, 2019-01-03:
> I can confirm more code in the wild which does an ifdef check on INPUT_PULLUP (and falls back to pinMode & digitalWrite), because INPUT_PULLUP was added to Arduino later than the others.

One way to fix that, without sacrificing the enum, is to add `#define INPUT_PULLUP INPUT_PULLUP`. I think we've used this trick for other macros converted to constants in the AVR core before (or at least discussed it).

As for the convert-from-boolean point, I guess one could introduce a class type to wrap these constants and add the appropriate conversion operators on them, though that might become a bit more complex than one would like. It does keep the advantage of separate types for each of these. It would also allow splitting PinStatus and SignalFlank (or whatever), and still allow using `LOW` and `HIGH` for both by adding implicit conversions.

Coding-Badly, 2019-01-05:
The _digitalRead_ can be handled with an operator...

```
bool operator ! (const PinStatus &v)
{
  return v == HIGH ? LOW : HIGH;
}
```

Coding-Badly, 2019-01-05:
The _digitalWrite_ can handled with an overload...

```
void digitalWrite( unsigned pin, PinStatus value )
{
...
}

void digitalWrite( unsigned pin, bool value )
{
  digitalWrite( pin, value ? HIGH : LOW );
}
```

Ideally, the _unsigned_ / _bool_ function is placed in a header file and marked to be inlined.

WestfW, 2019-02-07:
I agree with Floessie - having digitalRead/digitalWrite use pinStatus() is no better than the code that expects/passes booleans.    There is no reason to overload the enum with functionality from two separate sets of functionality...

per1234, 2019-02-07:
Today another report came in of a popular library broken by this:
https://bitbucket.org/fmalpartida/new-liquidcrystal

Minimal demonstration sketch:
```c++
#include <LiquidCrystal_I2C.h>
void setup() {}
void loop() {}
```
```
E:\arduino\libraries\Newliquidcrystal_1.3.5\LiquidCrystal.cpp: In member function 'virtual void LiquidCrystal::send(uint8_t, uint8_t)':

E:\arduino\libraries\Newliquidcrystal_1.3.5\LiquidCrystal.cpp:125:48: error: cannot convert 'bool' to 'PinStatus' for argument '2' to 'void digitalWrite(pin_size_t, PinStatus)'

    digitalWrite( _rs_pin, ( mode == LCD_DATA ) );

                                                ^

E:\arduino\libraries\Newliquidcrystal_1.3.5\LiquidCrystal.cpp: In member function 'void LiquidCrystal::writeNbits(uint8_t, uint8_t)':

E:\arduino\libraries\Newliquidcrystal_1.3.5\LiquidCrystal.cpp:305:48: warning: invalid conversion from 'int' to 'PinStatus' [-fpermissive]

       digitalWrite(_data_pins[i], (value >> i) & 0x01);

                                                ^

In file included from C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.6.210\cores\arduino/api/ArduinoAPI.h:52:0,

                 from C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.6.210\cores\arduino/Arduino.h:23,

                 from E:\arduino\libraries\Newliquidcrystal_1.3.5\LiquidCrystal.cpp:36:

C:\Users\per\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.6.210\cores\arduino/api/Common.h:104:6: note:   initializing argument 2 of 'void digitalWrite(pin_size_t, PinStatus)'

 void digitalWrite(pin_size_t pinNumber, PinStatus status);

      ^
```

Reported at:
http://forum.arduino.cc/index.php?topic=595987

facchinm, 2019-02-07:
Could something like this work for everyone? (it is from the `namespace_arduino` branch, hence the namespace stuff :slightly_smiling_face: 
https://github.com/arduino/ArduinoCore-API/blob/55cbc081bb5bcf1244743c3387b85ba56d0fe4b2/api/Compat.h#L1-L16

per1234, 2019-02-16:
@facchinm this is probably a dumb question, but it's not clear to me exactly how this would be implemented. I get the idea of overloading the functions. But `pinMode` and `digitalWrite` are C, which doesn't support function overloading in this manner.

I was trying to modify the Arduino megaAVR Boards core code to test this solution against the known issues reported so far in the thread. I tried just replacing the `api` folder with the one from the namespace_arduino branch of ArduinoCore-API, but then I get the same issues as well as some new unrelated errors.

facchinm, 2019-02-21:
@per1234 it's not a dumb question at all!
In the `namespace` branch all implementations are C++ only, and only one of them is declared `extern "C"` so it can be used from any C file.
In this case, the (non compact) implementation gets exposed, while the other one is only available in C++ files.
Maybe it's not the most elegant approach but it should solve 99% of the problems (if the nonstandard code lives in the sketch or in a c++ library)

per1234, 2019-02-21:
Thanks for the explanation! I agree that the amount of C code in existence affected by this issue is probably very small. I hadn't considered that the solution would be limited to C++/.ino.

MCUdude, 2019-04-20:
I'll have to agree with @WestfW and @Floessie on this one.

I understand that @facchinm tries to provide a solution that will keep the new implementation, but what do we need this new implementation in the first place? It only leaves ut with a workaround that's not compatible with C libraries, not any benefits as far as I can see.

JChristensen, 2019-06-26:
I'd be interested to know what problem is solved by introducing the PinStatus type. If this is about type safety, then I think there is such a thing as getting carried away. For example, it seems perfectly natural to pass a bool as the second argument to digitalWrite(). In that case, introducing PinStatus is not only confusing but seems like reinventing the wheel, as the language already provides a completely adequate type. If in fact this change is about type safety, then I'd have to wonder how big the problem was in the first place.

Perhaps worse, the current situation is inconsistent between various boards, with megaAVR architecture demanding a PinStatus type (e.g. for digitalWrite), and regular AVR being more permissive. In fact, PinStatus is not defined for the AVR architecture.

Coding-Badly, 2019-06-26:
Jack!  I hope you've been well!

Excellent question.  I can imagine two possibilities.

1. bool is severely limited in the amount of information that it can carry.  A pin status type can carry the same information and later, if necessary, be changed to carry additional information.  It's a way of "future proofing" the interface.

2. Having a type specific to pin status allows overloading.  It is possible for digitalWrite(pin, bool) to have different behaviour than digitalWrite(pin, PinStatus).  (I can't think of a valid reason to do that but I also have not put much thought into it.)

bxparks, 2019-07-01:
Question 1:

Looking at `$ARDUINO_IDE/portable/packages/arduino/hardware/megaavr/1.8.1/cores/arduino/wiring_digital.c`, I see that `digitalWrite()` now accepts 3 values of `PinStatus` (`HIGH`, `LOW` and `CHANGE`). It seems that `CHANGE` causes the pin to toggle, and CHANGE is defined to be an enum of 2.

However, on a normal AVR, CHANGE is defined to be 1 (see `$ARDUINO_IDE/hardware/arduino/avr/cores/arduino/Arduino.h`) so `digitalWrite(pin, CHANGE)` compiles, but has the same effect as `digitalWrite(pin, HIGH)`.

Is this intended?

Question 2:

Is there a `#define` macro that identifies the megaAVR boards which use this new `PinMode` and `PinStatus` API, so that I can fix my library for the megaAVR?

facchinm, 2019-07-02:
Hi @bxparks 

1) probably implementing CHANGE as pin toggle was not the best idea ever; anyway, not using it doesn't hurt :slightly_smiling_face: What's your use case?
2) you can target any core implementing PinMode and PinStatus with `#if (ARDUINO_API_VERSION >= 10000)`

bxparks, 2019-07-02:
Hi @facchinm,

I don't use `digitalWrite(pin, CHANGE)`... I was trying to deduce the valid range of `PinStatus` for  the new `digitalWrite()` and `digitalRead()`. Previously both of them were documented to handle only 2 values: `HIGH` and `LOW`. Now `digitalWrite()` accepts 3 values. So I'm wondering, will `digitalRead()` ever return any other values of `PinStatus`?

The reason for asking is that my [AceButton](https://github.com/bxparks/AceButton) library stores a value of `digitalRead()` that represents "UNKNOWN", in other words, the state of the pin before any `digitalRead()` has been performed. But the [documentation for digitalRead()](https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/) does not define the value of `HIGH` or `LOW`, so I have a compile-time check like this:
```C++
#if HIGH != 1
  #error HIGH must be defined to be 1
#endif
#if LOW != 0
  #error LOW must be defined to be 0
#endif
```
Which enforces that my "UNKNOWN" value is different from either `HIGH` or `LOW`. Obviously for the megaAVR, these checks no longer work. And obviously, I cannot use `PinStatus` to store my `digitalRead()` result, because "UNKNOWN" cannot be represented by the enum.

Thanks for the pointer to `ARDUINO_API_VERSION`. Is it correct to assume that eventually even the normal AVRs will migrate to the new API, with the `PinStatus` and `PinMode` enums?

facchinm, 2019-07-03:
The whole idea of HIGH and LOW is to hide the implementations details (HIGH could be 0 and LOW 0x12 in some random funky architecture, but your Arduino code will run seamlessly).
Expecting them to be 0 and 1 (with no other value allowed) breaks this assumption, so maybe there's a better way to enforce the UNKNOWN field :wink:

PaulStoffregen, 2019-07-03:
Overloading digitalWrite will break much of the Arduino ecosystem.  A tremendous amount of code has been built up over the last decade equating digitalWrite & digitalRead values with the C/C++ convention of zero representing logical false to mean LOW and non-zero representing logical true to mean HIGH.

facchinm, 2019-07-03:
@PaulStoffregen of course most (all?) the architectures are sane and HIGH is 1 while LOW is zero but since the APIs don't specify the value of those macros/enum/whatever (and never did) the code runs "by chance" :slightly_smiling_face:

bxparks, 2019-07-03:
I guess I'm having a hard time seeing what advantages the `PinStatus` API offers that outweighs the cost breaking backwards compatibility. We've already seen that the one new feature that it does provide, the ability to toggle using `digitalWrite(pin, CHANGE)` cannot be used, since it breaks compatibility with all other platforms.

I also understand that writing:
```C++
pinStatus = (pinStatus != LOW)? LOW : HIGH;
```
is probably more clear to most people than
```C++
pinStatus ^= 0x1;
```
but it would be nice to have the *option* of writing the second form in places where it makes sense (e.g. deep inside a library where app developers are not expected to visit often). The XOR can save a few bytes of flash memory and CPU cycles, since the compiler cannot optimize the first into the second. (Interestingly, when I wrote the first version using `?:` operator, I actually wrote it wrong initially and flipped the `LOW` and `HIGH`. That expression is not 100% easy to get correct. Hmm.)

With regards to `UNKNOWN`, yes it's always possible to use a second byte to store the `bool isUnknown` flag. But that's using 2 bytes to store 2 bits of information (LOW, HIGH, UNKNOWN), instead of just one byte. Now, if `PinStatus` contained a `PinStatus::UNKNOWN` whose value was guaranteed to be never returned by `digitalRead()`, then `PinStatus` would actually provide some value to me.

In any case, it looks like the new `PinStatus` API was introduced almost 2 years ago, so the ship has sailed? I will use
```C++
#if (ARDUINO_API_VERSION >= 10000)
...
#endif
```
to target different platforms appropriately.

WestfW, 2019-07-03:
>  the APIs don't specify the value of those macros/enum/whatever (and never did) the code runs "by chance"
wiring_shift.c in the arduino core ( https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring_shift.c#L32 ) passes a boolean to digitalWrite(), and the Firmata library uses 0/non-zero: https://github.com/firmata/arduino/blob/master/Boards.h#L970

PaulStoffregen, 2019-07-03:
That sort of code is pervasive throughout the Arduino ecosystem.

I fear these API changes could bring widespread misery.

bxparks, 2019-07-04:
I tried to protect myself against those assumptions with my compile-time checks, e.g.:
```C++
#if HIGH != 1
  #error HIGH must be defined to be 1
#endif
#if LOW != 0
  #error LOW must be defined to be 0
#endif
```
But even that breaks under this new API. :-/

WestfW, 2019-07-04:
That's presumably because HIGH and LOW are now part of an enum rather than mere constants.

You can probably do:
```
void unexpected_HIGH_or_LOW_value(void) __attribute__ (( error("") ));
// ...
void MyFunction (...) {
   if (HIGH != 1 || LOW != 0)
      unexpected_HIGH_or_LOW_value();
//    ...
}
```
Since HIGH and LOW are still compile-time constants, this won't normally produce any actual code, but they no longer need to be preprocessor-time constants.  (There are a bunch of places where it might be useful to use this sort of thing in the Arduino code.  But that's for another time/issue!)

bxparks, 2019-07-05:
(This is getting slightly off topic, but hopefully it's useful to someone else facing a similar problem with compile-time checking of `HIGH` and `LOW` values)

@WestfW: Thanks for the pointer! I didn't know about the `error` attribute. I discovered 3 problems with your solution unfortunately:
1) The compiler complains about an unused `MyFunction()` if this check is the only code inside.
2) I have to create a new `unexpected_HIGH_or_LOW_value()` function for each compile-time check that I want to perform.
3) The `error` attribute is a g++ only extension, clang++ does not recognize it. (This is a problem for me because I try to write portable code, so that I can compile and run my Arduino unit tests on Linux (g++ or clang++) and MacOS (clang++)).

After reading the [GCC guide on `__attributte__`](https://gcc.gnu.org/onlinedocs/gcc/Common-Function-Attributes.html) on the `extern char [(condition) ? 1 : -1];` trick, and [this blog on catching compile-time errors](https://www.embedded.com/electronics-blogs/programming-pointers/4025549/Catching-errors-early-with-compile-time-assertions), I came up with the following:
```C++
#define CONCAT_(x, y) x##y
#define CONCAT(x,y) CONCAT_(x,y)
#define COMPILE_TIME_ASSERT(cond, msg) \
    extern char CONCAT(compile_time_check, __LINE__)[(cond) ? 1 : -1];
COMPILE_TIME_ASSERT(HIGH == 1, "HIGH must be 1")
COMPILE_TIME_ASSERT(LOW == 0, "LOW must be 0")
```

Works on both a `#define HIGH` and an enum value of `HIGH`.

Avamander, 2020-01-29:
Can this be fixed as soon as possible? It's unnecessarily incompatible with a large majority of other platforms, `megaavr` is just an annoying outlier here.

Avamander, 2020-02-03:
@facchinm Can this god-awful change be reverted, there is really zero benefit to this breakage.

JAY-M-ERICSSON, 2020-04-03:
Part of me would say "too bad for those who never understood that type coherence is an important concept"... but I would vote against getting that live too. 

I've seen too much bad code that has been confusing a truth value (a bool) with a voltage status value (HIGH and LOW or even UNKNOWN) that I fear this change would have terrible impacts on newcomers trying to use random libraries and books that have been relying on the "expected" behavior. 

And the benefit is very limited...

MCUdude, 2020-04-06:
Since the devs aren't listening. I'm tired of this too, so I will get rid of this in the [MegaCoreX](https://github.com/MCUdude/MegaCoreX) repo. This core is 100% compatible with the Uno Wifi Rev2 and the Nano Every.

@SpenceKonde might do it for [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) as well.

facchinm, 2020-04-07:
@MCUdude[ this commit ](https://github.com/arduino/ArduinoCore-API/commit/9c2dbe25139cb68446b581ab505c0a4ab1b4bcdc) should solve all the issues about this topic (and is being used by mbed core, Nano 33 BLE, so there's some validation of the method).
The megaAVR core didn't receive any update in the meantime but I think we could release a new version of the core if needed.

MCUdude, 2020-04-07:
digitalRead is still not returning an int, but rather PinStatus. But this might be less of a problem.

> The megaAVR core didn't receive any update in the meantime but I think we could release a new version of the core if needed.

Yes, so much yes!

bperrybap, 2020-04-13:
@facchinm That commit solves the issue for digitalWrite() and pinMode() booleans, but it does not solve the issues related to code wanting to check for the existence of INPUT_PULLUP
i.e. I have a library that breaks with an error message because INPUT_PULLUP is not defined.
My library code explicitly checks for the existence of that macro.
Others noted this issue as well.

Common.h could resolve this by adding these defines just after the pinMode enum.

```
// create equivalant macro symbols for PinMode enums
#define INPUT INPUT
#define OUTPUT OUTPUT
#define INPUT_PULLUP INPUT_PULLUP
#define INPUT_PULLDOWN INPUT_PULLDOWN
```
Could we add this to Common.h

mohdrais, 2020-04-16:
@bperrybap Will it be safe to modify the common.h in the system? By the way,I cannot find the right common,h file in my computer, i.e C:\Program Files (x86)\Arduino\hardware\tools\avr\avr\include\avr\common.h

bperrybap, 2020-04-16:
@mohdrais , It should be.
This is what I have done for INPUT_PULLUP until it is fixed.
It isn't "common.h" but "Common.h" and it is down under the **packages** directory.
No clue where that is on Windows. On linux, with the current megaavr core release it is:
 ~/.arduino15/packages/arduino/hardware/megaavr/1.8.6/cores/arduino/api/Common.h

I added the macros just below the pinMode enum
```
// create equivalant macro symbols for PinMode enums
#define INPUT INPUT
#define OUTPUT OUTPUT
#define INPUT_PULLUP INPUT_PULLUP
#define INPUT_PULLDOWN INPUT_PULLDOWN
```
To provide full backward compatibility for the previously existing macros, there are other enums that should have symbols/macros created as well.

bxparks, 2020-08-07:
I maintain a handful of Arduino libraries which are moderately popular. I continue to get bug reports that my libraries don't work the MegaAVR, but they work for many other Arduino-like boards (AVR, ESP8266, ESP32, Teensy, SAMD). I don't have the time or motivation to debug the backwards incompatible breakages caused by the MegaAVR tool chain. I don't own a MegaAVR, nor do I plan on getting one in the future.

I am inclined to make the MegaAVR explicitly **unsupported** by my libraries. I'm curious to hear the opinions of any other other libraries maintainers who were/are bitten by these MegaAVR breakages.

SpenceKonde, 2020-08-07:
Instead of doing that, refer users to the  https://github.com/MCUdude/MegaCoreX which supports the ATmega4809 (including Uno WiFi Rev. 2 and Nano Every), and doesn't have this mess (I think it also adds a bunch of pwm pins, and does a variety of things better too). I have likewise reverted my https://github.com/SpenceKonde/megaTinyCore to do pinMode/etc the old way as well. Marking megaavr as unsupported architecture will also impact users of these third party cores on which they would otherwise work.

bxparks, 2020-08-07:
Ah ok, I'm not familiar with those other megaAVR cores (MegaCoreX and megaTinyCore). If they use the same old programming framework as all the other boards, then my libraries should work just fine with those. I guess I should narrow down my blacklist: How about I exclude the 2 entries I see under "Arduino megaAVR Boards", i.e. **Nano Every** and the **UNO WiFi Rev2**.

To be clear, I have nothing personal against those boards. But I maintain my libraries on my own time, for free, for fun. I see nothing fun in spending my limited time and patience, to track down random, unknown, undocumented, backwards incompatible breakages for these 2 megaAVR boards that I don't care about.

SpenceKonde, 2020-08-07:
I think that would work; I would also suggest a note in the README of the relevant libraries referring users of those boards to MegaCoreX (you can use it with the Nano Every/Uno WiFi Rev. 2 - you just tell the IDE it's a 4809, and it has a pin mapping option that matches the pin numbering on the officials board).

matthijskooijman, 2020-11-26:
I think that the suggestion from @bperrybap is actually good, just `#define` these values as themselves for compatibility. IIRC we already did the same thing for some other constant that was turned into a const or enum elsewhere (maybe in some Arduino core, or maybe some other library, don't remember exactly), and it seems to be a safe but effective way to allow `#ifdef` without messing up uses of these names in other contexts (i.e. it actually *fixes* some problems that occur when defining these constants directly when code tries to use the same name as a variable name, for example).

matthijskooijman, 2020-11-26:
Also, these defines should maybe be put in `Compat.h` rather than `Common.h` to mark them as deprecated and not recommended for use? For some thoughts about phasing out such compatibility stuff, see https://github.com/arduino/arduino-cli/issues/956

Michael-Brodsky, 2022-04-20:
You're absolutely right, the changes to the megaavr api broke tons of my code and I'm now having to sift through everything and put in conditionals. Nice of them to tell us, and there's no docs to speak of. Right on!!

bperrybap, 2022-04-20:
`You're absolutely right, the changes to the megaavr api broke tons of my code and I'm now having to sift through everything and put in conditionals. Nice of them to tell us, and there's no docs to speak of. Right on!!`

While I do agree that the documentation could be better,
most code that breaks is code that abused the digital i/o function API by making assumptions on the values of HIGH and LOW and the underlying implementation and used the digital pin i/o API in a way that was not assured/guaranteed by the existing minimal documentation.
i.e. the code did things in a way that was not supported by the existing documentation so it was not portable and got burned by abusing the API rather than use the API the way it was documented.

I do feel for the users that get caught in the middle.

The only thing that tripped up my library was the lack of a macro define for INPUT_PULLUP.
I got caught & burned by using something that was not documented.
i.e. checking for the existence of macro INPUT_PULLUP
I only did it in one diagnostic test example so it didn't affect the overall library or any other of its examples.

Since INPUT_PULLUP became part of Arduino in 1.0.1 ( the next release after 1.0)
I now no longer support Arduino 1.0 and will just let the one example ungracefully fail should INPUT_PULLUP not exist
This allows the code to run on all platforms that have the INPUT_PULLUP symbol regardless of how it is declared.
So far, I haven't run into a platform that doesn't have this symbol.

A much bigger concern for me that is related to this is that the Arduino IDE allows the tools in one platform to override the tools in another platform if the tools are newer and are for the same architecture.

This new behavior caused the installation of the megaavr platform to break other unrelated AVR platforms.
IMO, one platform should not be able to override something in another platform.
i.e. the installations of one platform should not be allowed to modify, override, or even break things in another installed platform.
Each platform should be separate and independent of the any other installed platform, even if they are the same architecture.

--- bill

bperrybap, 2022-04-20:
@Michael-Brodsky 
Yeah that sucks.
Were you around for the Arduino 1.0 release debacle? 
This stuff pinStatus & pinMOde API stuff hurts but nearly as bad as when they chose to intentionally break 100% of all the existing third party code & libraries when moving from the release candidate to the final release.

I'd "just say no" and drop support for the megaAVR platform until they fix a few things.
And don't forget that installing that platform can break other AVR platforms which can make the problem even worse.
In the earlier days of the megaAVR platform, I told my library users not only that that the library didn't support it but not to install it as it caused other issues.

While this github issue was about the PinStatus & pinMode stuff, some of the issues I'm seeing in this github issue are do to some of the other things that have been done, or not done,  in the megaavr platform beyond the PinStatus and pinMode types.

I'm still not sure of the value of using stricter types like enums in the API at this point in time.
Yeah it would have made sense 10+ years ago, but at this point in time it seems to have much more downside than upside, given the desire/need for a certain level of backward compatibility for existing code.
There are now function overloads to get around having to use the new types so what value do the new types really bring to the table even for the long term?

IMO, the only way out of these types of issues would have been to plan for API migration and make these types of significant changes with a new Arduino version like say 2.0
And then have an option in the IDE GUI to set a compile time define to enable pre 2.0 API compatibility to allow existing code to continue to work for some period of time - which might be forever.

Michael-Brodsky, 2022-04-21:
Hey Bill, 

No I wasn’t around for v 1.0. I just started this a couple few years ago, developed some simple industrial automation apps to see how they’d work https://youtu.be/fCEVc_z0KCA. Turns out I had quite a bit of success  and have kept at it. My current project is a sorta open source version of LabJack on an Arduino along with a Windoze desktop app that forms the basis of an HMI/SCADA system. I hope to post a demo on YT this coming weekend.

The problems appeared when I bought a WiFi Rev 2 to demo the system’s over-the-air performance and everything broke. I’ve managed to get everything working, but I have a lot of time invested in this project and hunting down api bugs throws me off my train of thought and the problem at ha
I’ve raised these issues on the Arduino Forums with about zero luck. You’re actually the first person I’ve met that has an understanding of the issues. I like some of your ideas like the overloads thing or even going further to templates. As long as it stayed within the realm of template argument deduction, users wouldn’t see a difference. 

Anyway, glad to have run into you. Would like to stay in touch and discuss this and other topics further. 

Cheers

bxparks, 2022-04-28:
For what it's worth, I decided to explicitly blacklist the ArduinoCore-API for most the 15-20 Arduino libraries that I publish publicly, using something like the following at the top of the main header file:

```C++
#if defined(ARDUINO_API_VERSION)
#error Platforms using ArduinoCore-API not supported
#endif
```

One, it gives a short, understandable error message to the end-user, instead of hundreds of lines of inscrutable compiler errors.

Two, it completely stopped the repeated bug tickets against my libraries, saying "Your libraries does not work with [MegaAVR | Nano Every | RaspberryPi Pico | MKR | etc...]".

Three, it saved me dozens (maybe hundreds?) of hours of development and debugging time that I don't have to spend porting my libraries to the new Arduino API, and hundreds of dollars for buying the various microcontrollers that I no longer have to test and validate.

PaulStoffregen, 2022-04-28:
FWIW, I'm keeping an eye on this with concern about how this is likely to affect the Arduino ecosystem, and eventually what I'll need to support on Teensy boards.

Michael-Brodsky, 2022-05-07:
Hi guys,

For what it's worth, after hunting down all the relevant megaavr bugs and
refactoring my code to get it working, I compiled a list of resolved and
persistent issues. Just providing these in case they're useful to someone.

This is a list of issues encountered using the megaavr architecture and my
current fixes:

Problem 1: Bad or missing C Standard Library header <string.h>
Solution: Wrote my own that includes about a dozen of the functions that my
libs/apps depend on (might be ANSI/ISO compliant, maybe, kinda, sorta).

Problem 2: Macro definitions of "pin states" (LOW,HIGH,FALLING,RISING,etc.)
redefined as an enumerated type `PinStatus' in
..\arduino\hardware\megaavr\1.8.7\cores\arduino\api\Common.h
Solution: Conditionals to detect architecture, `using PinStatus = int;' for
non-megaavr architectures, and lots of `static_cast<PinStatus>(...)'.

Problem 3: avr WiFi api generates errors/doesn't compile when used
according to Arduino example sketches.
Solution: Dependencies are sensitive to the order in which they're
included.

Problem 4: Signature of WiFi::begin(...) function changed ssid parameter
from type char* to const char*.
Solution: Conditional typedefs.

For both problems 3 & 4, I wrote a header to take care of the dependency
order and typedefs:
# include <SPI.h>
# if defined ARDUINO_ARCH_MEGAAVR // megaavr uses WiFiNINA:
#  include <WiFiNINA.h>
#  include <WiFiUdp.h>
using ssid_t = const char*; // WiFiNINA ssid param is type const char*.
# else // everything else uses whatever these are,
#  include <WiFi.h> // and they must be included in this order:
#  include <WiFiClient.h>
#  include <WiFiServer.h>
#  include <WiFiUdp.h>
using ssid_t = char*; // everything else ssid param is type char*.
# endif // defined ARDUINO_ARCH_MEGAAVR

The following is a list of persistent issues on both avr and megaavr
architectures. They're compiler/uploader warnings that seem to all be
related to the WiFi and Ethernet api's.

***AVR and MEGAAVR***:
In file included from C:\Program Files
(x86)\arduino-1.8.16\libraries\Ethernet\src\Dns.cpp:8:0:
C:\Program Files (x86)\arduino-1.8.16\libraries\Ethernet\src\Dns.cpp: In
member function 'uint16_t DNSClient::BuildRequest(const char*)':
C:\Program Files
(x86)\arduino-1.8.16\libraries\Ethernet\src\utility/w5100.h:457:25:
warning: result of '(256 << 8)' requires 18 bits to represent, but 'int'
only has 16 bits [-Wshift-overflow=]
 #define htons(x) ( (((x)<<8)&0xFF00) | (((x)>>8)&0xFF) )
                      ~~~^~~

C:\Program Files
(x86)\arduino-1.8.16\libraries\Ethernet\src\Dns.cpp:164:18: note: in
expansion of macro 'htons'
  twoByteBuffer = htons(QUERY_FLAG | OPCODE_STANDARD_QUERY |
RECURSION_DESIRED_FLAG);
                  ^~~~~

***AVR only***:
In file included from C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src/WiFi.h:26:0:
C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src/utility/wl_definitions.h:42:0:
warning: "MAX_SOCK_NUM" redefined
 #define MAX_SOCK_NUM  4

C:\Program Files
(x86)\arduino-1.8.16\libraries\Ethernet\src/Ethernet.h:39:0: note: this is
the location of the previous definition
 #define MAX_SOCK_NUM 8

C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src\utility\wifi_drv.cpp: In static
member function 'static uint8_t WiFiDrv::getEncTypeNetowrks(uint8_t)':
C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src\utility\wifi_drv.cpp:451:10:
warning: converting to non-pointer type 'uint8_t {aka unsigned char}' from
NULL [-Wconversion-null]
   return NULL;
          ^~~~

C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src\utility\wifi_drv.cpp: In static
member function 'static int32_t WiFiDrv::getRSSINetoworks(uint8_t)':
C:\Program Files
(x86)\arduino-1.8.16\libraries\WiFi\src\utility\wifi_drv.cpp:476:10:
warning: converting to non-pointer type 'int32_t {aka long int}' from NULL
[-Wconversion-null]
   return NULL;
          ^~~~

***MEGAAVR only***
Uploader generates the following warning:

avrdude: WARNING: invalid value for unused bits in fuse "fuse5", should be
set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xcd instead of 0xc9 (double check with your
datasheet first).

Cheers!

Michael B.

Michael-Brodsky, 2022-05-08:
Here's some suggestions to fix the `bit shift overflow' and 'duplicate
definition' warnings:

// TYPE INFERENCE

template<class T>
auto bitShiftThingy(T x)  -> decltype(((x << 8) & 0xFF00) | ((x >> 8) &
0xFF))
{
  return ((x << 8) & 0xFF00) | ((x >> 8) & 0xFF);
}

// ENCAPSULATED COMPILE-TIME CONSTANTS

class EtherMonster
{
public:
  static constexpr uint8_t MaxSockNum = 8;

};

class WiFiGuy
{
public:
  static constexpr uint8_t MaxSockNum = 4;

};

This compiles without errors or warnings (it does however require 64-bit
type support for the println() function if you want to print intmax_t. I
use one in my libs.)

[image: Arduino IDE.png]

drf5n, 2023-03-24:
I like that the PinStatus enum names LOW = 0 and HIGH = 1 as used in the text of :

https://docs.arduino.cc/built-in-examples/basics/DigitalReadSerial

from 

https://github.com/arduino/docs-content/blob/main/content/built-in-examples/01.basics/DigitalReadSerial/DigitalReadSerial.md

>Now that your setup has been completed, move into the main loop of your code. When your button is pressed, 5 volts will freely flow through your circuit, and when it is not pressed, the input pin will be connected to ground through the 10k ohm resistor. This is a digital input, meaning that the switch can only be in either an on state (seen by your Arduino as a "1", or HIGH) or an off state (seen by your Arduino as a "0", or LOW), with nothing in between.
...
>Now, when you open your Serial Monitor in the Arduino Software (IDE), you will see a stream of "0"s if your switch is open, or "1"s if your switch is closed.

and the 

         thirdSensor = map(digitalRead(2), 0, 1, 0, 255);

line from https://docs.arduino.cc/built-in-examples/communication/SerialCallResponse

It doesn't seem fair to call depending on LOW=0/HIGH=1 as bad code if the reference documentation has been pushing it for years

On the other hand, isn't this PinStatus enum inconsistent with the either-or text from https://www.arduino.cc/reference/en/language/functions/digital-io/digitalread/ and 
https://www.arduino.cc/reference/en/language/functions/digital-io/digitalwrite/

And doesn't this break expectations for this code:

https://github.com/arduino/ArduinoCore-megaavr/blob/ffab9cb2e4bca7647f11d6e25727788aec597a03/cores/arduino/wiring_digital.c#L148-L161

If you call `digitalWrite(13,FALLING);` should it set the pin or clear the pin?

SpenceKonde, 2023-03-24:
IMO this whole concept of pin enums was something that would be great if we were starting from scratch low end ARMs where we could afford to piss away some flash like that. But we''er on AVRs, there's a long established tradition of the 0/1 being legal for digitalWrite, and it's easy to come up with plausible use cases and common ideoms where it falls over. 

That's why I 86'ed it from my cores. (the whole Arduino API was sort of a disaster IMO - at this point it's been pretty much gutted on my cores as people have complained (with good reason) about the bloat it introduced (virtual functions are the devil's creation).

WestfW, 2023-03-25:
If we were starting from scratch, "pins" should probably have been objects.

aentinger, 2023-03-27:
>virtual functions are the devil's creation

Why are virtual functions the devils creation? Please inform me about the performance cost you think they'll incur ;)

bperrybap, 2023-03-27:
One issue with virtual functions is that they will get linked in regardless if they are used.
The compiler & linker options that are used to remove unused functions don't work for virtual functions and while there was  a way (hack) to patch the vtable to allow the linker to remove unused fucntions, that no longer works in the more recent gcc toolsets.

ianfixes, 2023-05-02:
I'm only just running into this issue in [Arduino CI](https://github.com/Arduino-CI/arduino_ci); I found this thread by running a google search for the actual definition of `digitalWrite`, and so far this discussion is very discouraging.

Arduino CI enables unit testing an Arduino library by mocking the entire set of Arduino built-ins, such that they can be fed scripted inputs and have their outputs analyzed.  See [the definitions of `HIGH` and `LOW`](https://github.com/Arduino-CI/arduino_ci/blob/6f326f8070e619802ff6efe07bc3340313806d1f/cpp/arduino/ArduinoDefines.h#L5-L6) and [`digitalRead`/`digitalWrite`](https://github.com/Arduino-CI/arduino_ci/blob/6f326f8070e619802ff6efe07bc3340313806d1f/cpp/arduino/Godmode.h#L166-L167) respectively.

To make code more testable, I prefer to follow the [dependency injection](https://allencch.wordpress.com/2017/12/05/c-unit-test-and-dependency-injection/) pattern.  This simplifies the problem of testing, since I can simply write a mocked version of `digitalRead` and use it to precisely simulate the inputs I want my library to handle.  All I need to do is declare the precise type of the `digitalRead` function.

### But what is the correct type signature of this function???

For this to have 2 different signatures for 2 different boards spells disaster for any library that wants to work for multiple boards -- it looks like I have to pick one signature or the other, and using preprocessor directives to sort this out seems like a very unsustainable way to go.

How can I encapsulate the possibly changing signature of this function in (1) an Arduino library that I write and (2) my mocks for testing Arduino libraries as part of Arduino CI?

ianfixes, 2023-05-02:
Addendum, even overloading isn't an obvious solution here, because in some cases `PinStatus` won't even be defined.

bperrybap, 2023-05-02:
> For this to have 2 different signatures for 2 different boards spells disaster for any library that wants to work for multiple boards -- it looks like I have to pick one signature or the other, and [...]

Number 1 rule: never abuse the API and use the APIs the way they are documented.
(using a specific type for HIGH and LOW is abusing the API)
You have to treat HIGH and LOW as an opaque unknown type.
- never assume that their type is anything and could change.
- never attempt to assign HIGH or LOW to a variable and never use a variable as the value parameter to digitalWrite()

You do things like when you call functions like digitalWrite() always use HIGH or LOW for the value parameter, not a variable.

For digitalRead() always compare the return value to LOW or HIGH and then do things based on that evaluation.
i.e. if LOW set a variable to some value, if HIGH then set it to some other value.
When using your variable assigned based on return value of digitalRead() to set a pin using digitalWrite(), look at the variable and then call dititalWrite() with LOW or HIGH 

My libraries do this and it didn't have any issues with the different types for HIGH and LOW.
It works on all boards within all cores and the types for HIGH and LOW can change and it won't affect my code compiling and working.

If you are trying to do something special or out of the ordinary for testing / emulation purposes, again you cannot ever assume any type for the symbols HIGH and LOW given the way the API is documented and defined.

If you for some reason absolutely need their type you can get to them  by using decltype()
i.e decltype(HIGH) or decltype(LOW)

So I would think that  between a clever combination of use of macros to remap function names and the use decltype() you could get to anything you want / need for your simulation / emulation purposes.

stuff like
```
decltype(HIGH) my_digitalRead(int pin);
void my_digitalWrite(int pin, decltype(HIGH));
decltype(HIGH) value; // this variable can be used with digitalRead() and digitalWrite()

```

You could even create a typedef for the decltype(HIGH) that would hide this away

Michael-Brodsky, 2023-05-02:
Untyped object-like macros, IMHO this is half the problem with the API.

SpenceKonde, 2023-05-02:
Thanks for the tip on decltype() 

@Michael-Brodsky hits the nail on the head - #define was horribly overused throughout the API. And where things were typed, int's were overused and uint8_t's underused.... **But now it is too late to fix most of this stuff** - there's too much code in the wild, and we long ago reached the point where the API was set in stone effectively. We can't go changing it without destroying compatibility short of straight-up magic, where one could shout "abracadabra!"* while brandishing an illuminated, arduino powered wand, and have all the code in existence changed to match a new API

*I've been told that this was originally an invocation of the demon-god of numbers and math; I don't know enough about such mythology to know if that's true or not, but that's the demon upon whose power you'd need to call to retroactively correct an API. Based on the absence of anyone working such magic, I conclude that the demon must demand too high a price for such alterations of reality (I certainly would if I were such a demon - it looks like a lot of work, even with unholy supernatural powers you'd expect of that sort of entity). It could also be that demons like that don't exist, and we'd need a full on time machine, and physics has pretty much constrained those out of the realm of possibility (well, for backward time travel. Forwards time travel is easy. I just used a time machine to travel forwards a half hour writing this crap)

bperrybap, 2023-05-02:
While the Arduino core APIs in general have quite a few issues, IMO, it goes back to the APIs being originally defined by the early Arduino.cc dev team which from my observations appeared to have little real world development and coding experience particularly in C++, but the real breakage in cases like this is due to so many people abusing the APIs by making assumptions that just are not in the API documentation,
And while many of the abuses and assumptions about the API are reasonable, they are not defined in the API so they still abuse the API to create non portable code.
i.e. users essentially wrote non portable code - at least according to the limited documentation.

Not defending Arduino.cc's choices, just saying the current state of things has been caused by a combination of issues on both sides.
(Arduino.cc APIs, documentation, their example coding choices and Arduino users coding choices)

About the only thing Arduino.cc could do now is to either accept the most common abuses and document the APIs as actually working that way (which would be my choice)
or
have an IDE option to allow building in a "strict" type mode for the newer types or a compatibility mode to allow the code that abuses the API to still continue to function.
or
come up with a different new & better set of APIs where you have the freedom to make them better based on the 15+ years of Arduino user experiences.

Michael-Brodsky, 2023-05-02:
What about incrementally working up to an ISO compliant version of the API,
while still maintaining backwards compatibility? That’s one of the reasons
I spent so much time implementing my std lib, that and necessity. When all
else fails, I just use wrappers that meet the ISO standards. Just an idea.

[6 more replies]