arduino/ArduinoCore-API #36: add pin open-drian and push-pull mode

Pillar1989, 2019-07-01:
https://github.com/arduino/ArduinoCore-API/blob/e1eb8de126786b7701b211332dda3f09aa400f35/api/Common.h#L18-L23
```
typedef enum {
  INPUT           = 0x0,
  OUTPUT          = 0x1,
  INPUT_PULLUP    = 0x2,
  INPUT_PULLDOWN  = 0x3,
  OUPUT_PUSHPULL  = 0x4,
  OUPUT_OPENDRAIN  = 0x5,
} PinMode;
```

PaulStoffregen, 2019-07-01:
What does OUPUT_PUSHPULL mean?  How is it different from regular OUTPUT?

Pillar1989, 2019-07-01:
https://en.wikipedia.org/wiki/Push%E2%80%93pull_output 

It seems that the basic circuit knowledge of this kind of chip still needs to be popularized.

PaulStoffregen, 2019-07-01:
Maybe you didn't understand my question.  The normal OUTPUT name Arduino has supported since its earliest days has represented the normal push-pull output circuitry.

So the question is *why* do we need the name OUPUT_PUSHPULL which represents the exact same thing?

Pillar1989, 2019-07-02:
@PaulStoffregen  You are right. I am really sloppy on this issue. Is the definition below better?
```
typedef enum {
  INPUT           = 0x0,
  OUTPUT          = 0x1,
  OUTPUT_PUSHPULL = 0x1,
  INPUT_PULLUP    = 0x2,
  INPUT_PULLDOWN  = 0x3,
  OUPUT_OPENDRAIN  = 0x4,
} PinMode;
```