arduino/ArduinoCore-API #39: Only one pin interrupt on nRF52840 devices

ohazi, 2019-09-01:
Attaching interrupts for multiple pin change events via attachInterrupt(...) doesn't appear to work on the new nRF52840 based nano boards. 

In the simplified test case below, the first call to attachInterrupt appears to work (falling edge of IN3, or pin 6). The other ones don't work. You can change the order of the attachInterrupt calls to get a different pin change interrupt to start working, but I've only ever seen one interrupt working at a time. 

I think pins 4, 5, and 6 are actually pins P1_15, P1_13, and P1_14 on the nRF52840 (from variant.cpp), although this could be wrong (the documentation for these new nano boards is pretty lacking). Could this be because all of the inputs are on the same port? This normally isn't a restriction, e.g. when using Nordic's SDK, although I haven't used mbed on a Nordic IC. 

In any case, this seems like a bug. 

```
#define IN1 4
#define IN2 5
#define IN3 6

// LEDs
#define OUT1  22
#define OUT2  23
#define OUT3  24

int out1 = 1;
int out2 = 1;
int out3 = 0;

void in1_handler() {
  out1 = (out1 + 1) % 2;
  digitalWrite(OUT1, out1);
}

void in2_handler() {
  out2 = (out2 + 1) % 2;
  digitalWrite(OUT2, out2);
}

void in3_handler() {
  out3 = (out3 + 1) % 2;
  digitalWrite(OUT3, out3);
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  while(!Serial) {}
  Serial.println("setup");

  digitalWrite(OUT1, out1);
  digitalWrite(OUT2, out2);
  digitalWrite(OUT3, out3);

  pinMode(OUT1, OUTPUT);
  pinMode(OUT2, OUTPUT);
  pinMode(OUT3, OUTPUT);

  pinMode(IN1, INPUT_PULLUP);
  pinMode(IN2, INPUT_PULLUP);
  pinMode(IN3, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(IN3), in3_handler, FALLING);
  attachInterrupt(digitalPinToInterrupt(IN2), in2_handler, FALLING);
  attachInterrupt(digitalPinToInterrupt(IN1), in1_handler, FALLING);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("delay");
  delay(500);
}
```

ohazi, 2019-09-01:
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/7

ohazi, 2019-09-01:
https://github.com/ARMmbed/mbed-os/issues/10949

sandeepmistry, 2019-09-03:
Closing this in favour of https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/7, which is the correct place for the issue.

closed 2019-09-03 by sandeepmistry