arduino/ArduinoCore-API #51: map() function equation wrong

st42, 2019-09-16:
EDIT NOTE: I added a comment to the end of this from a later comment because I thought it should have been included in the first place.

The equation in the function stated at http://arduino.cc/en/Reference/Map

``` C++
long map(long x, long in_min, long in_max, long out_min, long out_max)
{
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}
```

is wrong. The example given:

``` C++
val = map(val, 0, 1023, 0, 255);
```

In this example 0-1023 is mapped to 0-255. The equation gets it wrong by mapping 1023 numbers to 255 numbers, when it should be mapping 1024 numbers to 256 numbers.

0 through 1023 is 1024 numbers not 1023 : in_max - in_min : 1023-0=1023
0 through 255 is 256 numbers not 255 : out_max - out_min : 255-0=255

The corrected equation in the function should be:

``` C++
long map(long x, long in_min, long in_max, long out_min, long out_max)
{
  return (x - in_min) * (out_max - out_min + 1) / (in_max - in_min + 1) + out_min;
}
```

In the following simpler example it will be easier to see:

val = map(val, 0, 3, 0, 1);

In this example 0,1,2 and 3 is mapped to 0 and 1. Mapped correctly it should be:

0 maps to 0
1 maps to 0
2 maps to 1
3 maps to 1

Using the equation in the the function as it is now, it is mapped as such:

0 maps to 0
1 maps to 0
2 maps to 0
3 maps to 1

This is a common problem programmers have dealt with before. When dealing with arrays you can not determine array size by simply subtracting first position from last position you will always be off by 1. That is why you add 1 to get proper size. I know this is for the most part a minor problem, but if your project requires accurate results this could be a serious issue.

--- Added from later comment---

To give a real world scenario as to why this needs changed consider this. You are using the arduino with a sensor to sample data and based on this data set a servo into 1 of 4 positions. The arduino uses a 10 bit adc and outputs integers from 0 to 1023. You would use map(x, 0, 1023, 1, 4). With the map function as it is currently the only time position 4 would be active is when the arduino outputs the integer 1023. With my proposed change each position would get an equal share of the possible range. When they converted this function from a float function to an integer function they didn't consider that the output is no longer a value but a position. Value mapping maps one value to another and the function works perfectly for floats. Positional mapping distributes the input as evenly as possible to the output and the function as is doesn't do that.

This function needs to be split into two functions. One for floats as it is now and one for integers as I proposed.

PaulStoffregen, 2020-10-31:
I recently created an integer-only map() function which (I believe) solves all the problems @cmaglie mentioned above (over 5 years ago), including the "strange behaviour with negative numbers".

It matches the results using floating point and with proper round off to the nearest integer.

When used "backwards" it gives identical results (mapping 20-0 to 4-0 should be identical to mapping 0-20 to 0-4) and when used in "reverse" (mapping 0-20 to 0-4 compared with 0-20 to 4-0) gives the same set of results in reverse order.

The only downside is the algorithm is quite complicated and slow....

``` c
long idealMap(long x, long in_min, long in_max, long out_min, long out_max) {
  long in_range = in_max - in_min;
  long out_range = out_max - out_min;
  if (in_range == 0) return out_min + out_range / 2;
  // compute the numerator
  long num = (x - in_min) * out_range;
  // before dividing, add extra for proper round off (towards zero)
  if (out_range >= 0) {
    num += in_range / 2;
  } else {
    num -= in_range / 2;
  }
  // divide by input range and add output offset to complete map() compute
  long result = num / in_range + out_min;
  // fix "a strange behaviour with negative numbers" (see ArduinoCore-API issue #51)
  //   this step can be deleted if you don't care about non-linear output
  //   behavior extrapolating slightly beyond the mapped input & output range
  if (out_range >= 0) {
    if (in_range * num < 0) return result - 1;
  } else {
    if (in_range * num >= 0) return result + 1;
  }
  return result;
}
```

I also wrote of this on the Arduino forum.
https://forum.arduino.cc/index.php?topic=711016.0

If anyone wants to compare them all, here's a sketch you can run on Arduino Uno or other boards to see them all side-by-side.

``` c
const long in1 = 0;
const long in2 = 20;
const long out1 = 0;
const long out2 = 4;

void setup() {
  Serial.begin(9600);
  while (!Serial) ; // wait for Arduino Serial Monitor
  delay(100);
  Serial.println("map() function test");

  printHeader(in1, in2, out1, out2);
  for (long i = in1 - 10; i <= in2 + 10; i++) {
    printLong(i, 4);
    printLong(map(i, in1, in2, out1, out2), 7);
    printLong(map(i, in2, in1, out2, out1), 5);
    printLong(st42map(i, in1, in2, out1, out2), 7);
    printLong(st42map(i, in2, in1, out2, out1), 5);
    printLong(roundMap(i, in1, in2, out1, out2), 7);
    printLong(roundMap(i, in2, in1, out2, out1), 5);
    printLong(idealMap(i, in1, in2, out1, out2), 7);
    printLong(idealMap(i, in2, in1, out2, out1), 5);
    Serial.print("    ");
    Serial.println(floatMap(i, in1, in2, out1, out2));
  }
  printHeader(in1, in2, out2, out1);
  for (long i = in1 - 10; i <= in2 + 10; i++) {
    printLong(i, 4);
    printLong(map(i, in1, in2, out2, out1), 7);
    printLong(map(i, in2, in1, out1, out2), 5);
    printLong(st42map(i, in1, in2, out2, out1), 7);
    printLong(st42map(i, in2, in1, out1, out2), 5);
    printLong(roundMap(i, in1, in2, out2, out1), 7);
    printLong(roundMap(i, in2, in1, out1, out2), 5);
    printLong(idealMap(i, in1, in2, out2, out1), 7);
    printLong(idealMap(i, in2, in1, out1, out2), 5);
    Serial.print("    ");
    Serial.println(floatMap(i, in1, in2, out2, out1));
  }
}

float floatMap(float x, float in_min, float in_max, float out_min, float out_max) {
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

long idealMap(long x, long in_min, long in_max, long out_min, long out_max) {
  long in_range = in_max - in_min;
  long out_range = out_max - out_min;
  if (in_range == 0) return out_min + out_range / 2;
  // compute the numerator
  long num = (x - in_min) * out_range;
  // before dividing, add extra for proper round off (towards zero)
  if (out_range >= 0) {
    num += in_range / 2;
  } else {
    num -= in_range / 2;
  }
  // divide by input range and add output offset to complete map() compute
  long result = num / in_range + out_min;
  // fix "a strange behaviour with negative numbers" (see ArduinoCore-API issue #51)
  //   this step can be deleted if you don't care about non-linear output
  //   behavior extrapolating slightly beyond the mapped input & output range
  if (out_range >= 0) {
    if (in_range * num < 0) return result - 1;
  } else {
    if (in_range * num >= 0) return result + 1;
  }
  return result;
}

long st42map(long x, long in_min, long in_max, long out_min, long out_max)
{
  if ((in_max - in_min) > (out_max - out_min)) {
    return (x - in_min) * (out_max - out_min + 1) / (in_max - in_min + 1) + out_min;
  } else {
    return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
  }
}

long roundMap(long x, long in_min, long in_max, long out_min, long out_max)
{
  return ((x - in_min) * (out_max - out_min) + (in_max - in_min) / 2) / (in_max - in_min) + out_min;
}

void loop() {
}
```

bperrybap, 2021-04-06:
Here is one I did and proposed MANY years ago, when there was a long heated discussion on this topic.
```
long newmap(long x, long in_min, long in_max, long out_min, long out_max)
{
	if( x == in_max)
		return out_max;
	else if(out_min < out_max)
		return (x - in_min) * (out_max - out_min+1) / (in_max - in_min) + out_min;
	else
		return (x - in_min) * (out_max - out_min-1) / (in_max - in_min) + out_min;
}
```

bperrybap, 2021-04-06:
Here is a test sketch that can be run on an Arduino to test range sets.
The code includes 4 different map() functions for comparison.
map() - the one in the platform being compiled for
maprnd() - one that tries to do rounding - was proposed in the initial issue post
mapESP() - the one that comes with the ESP8266 platform
newmap() - one that I purposed years ago

To use,  modify the variables:
fromLow, fromHigh, toLow, toHigh

Then compile and upload to your Arduino and look at the serial port output.
Here are some interesting ranges
0,1023 to 0,100  the stock map() function doesn't round properly on the very high end of the from range
0,1023 to 0,255 (same as above)
1,3 to 1,5  maprnd() and mapESP() severely fail for this range
0,10 to 10,0  maprnd() and mapESP() severely fail for this range
-10,0 to 10,0 maprnd() and mapESP() severely fail for this range
0,100 to 0,255 maprnd() fails at the high end of the from range as it cannot get to 255 for the output

here is the code:
```
#if 0
// this is the map() function that is currently in the Arduino.cc core library
long map(long x, long in_min, long in_max, long out_min, long out_max)
{
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}
#endif

// a "rounding" version that some suggest
long maprnd(long x, long in_min, long in_max, long out_min, long out_max)
{
        return (x - in_min) * (out_max - out_min+1) / (in_max - in_min+1) + out_min;
}

// this is the map() function that is currently in the ESP866 platform core library
long mapESP(long x, long in_min, long in_max, long out_min, long out_max) {
    const long dividend = out_max - out_min;
    const long divisor = in_max - in_min;
    const long delta = x - in_min;

    return (delta * dividend + (divisor / 2)) / divisor + out_min;
}

// bperybap version
long newmap(long x, long in_min, long in_max, long out_min, long out_max)
{
	if( x == in_max)
		return out_max;
	else if(out_min < out_max)
		return (x - in_min) * (out_max - out_min+1) / (in_max - in_min) + out_min;
	else
		return (x - in_min) * (out_max - out_min-1) / (in_max - in_min) + out_min;
}

long fromLow = 0;
long fromHigh = 10;
long toLow = 10;
long toHigh = 0;

void setup(void)
{
long val, mval, mval2, mvalESP, mvalnew;
long dir = 0;
char buf[128];

	Serial.begin(115200);
	while(!Serial){}
	delay(500);
	Serial.print("\n\n");

	if(fromLow < fromHigh)
		dir = 1;
	else
		dir = -1;

	for( val = fromLow;; val += dir)
	{
		mval =       map(val,fromLow,fromHigh, toLow,toHigh);
		mval2 =     maprnd(val,fromLow,fromHigh, toLow,toHigh);
		mvalESP = mapESP(val,fromLow,fromHigh, toLow,toHigh);
		mvalnew = newmap(val,fromLow,fromHigh, toLow,toHigh);
		

		sprintf(buf, "map(%3ld,%3ld,%3ld,%3ld,%3ld)",
		 val, fromLow, fromHigh, toLow, toHigh);

		Serial.print(buf);

		sprintf(buf, " map:%3ld, maprnd:%3ld, mapESP:%3ld, newmap:%3ld\n",
		 mval, mval2, mvalESP, mvalnew);

		Serial.print(buf);

		if(val == fromHigh)
			break;
	}
}
void loop(void){}
```

per1234, 2021-06-19:
There is additional discussion on the subject at https://github.com/arduino/Arduino/issues/10382, which I am closing as a duplicate of this one.

bperrybap, 2021-06-19:
So why can't we fix this with either my proposed map function or Paul's templates as either do the mappings correctly (the way people would expect) and work for some cases where others fall flat on their face ad screw up for certain ranges like the current Arduino.cc IDE supplied map() function, the map() function proposed in the first post in this issue, and the ESP8266 core map() function.

IMO, I think Paul's template is the way to go as it works properly and as expected for more than just ints.
C++ templates are ideal for this kind of stuff  as long as it is ok to no longer have a map() function for C code.

How big of deal is it that fixing the existing map() function with better code that works more correctly potentially breaks some small amount of existing code that has attempted to work around the issue by fudging up the parameters?
And is support for C important or can it just exist and work with C++

It seems like we have a working solution for the calculation and it is now a question of deciding on the details of how to get it integrated.
For example, does it replace the existing map() function, is a a new function with a different name, or should it be handled as method function off of some "math" or "map" object ? and that object might be some sort of math class library
And is support for C important or can it just exist and work with C++

The reason I bring up a math or map object is that this math object could be used for all kinds of Arduino math functions that also have issues.
i.e. all the functions / macros that overlaying and overriding standard C library functions.
By supplying an Arduino "math" library with a default math object with templates it could fix several things at once
like map(), min(), max() etc....
by using the new method functions like math.map() math.min(), math.floor(), etc....
It also avoids any backward compatibility issues.

Seems like once we decide these questions, a solution could be integrated into the IDE fairly easily / quickly.

PaulStoffregen, 2021-06-20:
I'm pretty sure the answer to "why can't we" comes down to Arduino's priorities as a company.
[image]