arduino/ArduinoCore-API #76: round() macro in Arduino.h

Koepel, 2019-09-16:
The function round() in "math.h" is okay, it returns a floating point number as expected.
The macro round() in "Arduino.h" is a bug. It can not handle large floating point numbers and it returns a long integer. 
`#define round(x) ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))`
It could be renamed to something else, for example ROUND_INT().

The sketch below shows the difference. With the normal round() function it returns four times "1.50", and with the round() macro in "Arduino.h" it returns: "1.00", "1.50", "1.50", "0.00".

```
// Arduino IDE 1.81 with Arduino Uno

// The "#undef round" removes the round() macro in Arduino.h
// Try with and without the #undef round
// #undef round

void setup() 
{
  float x;
  Serial.begin( 9600);

  x = 15.0;
  x = round( x) / 10;   // a integer division with the macro
  Serial.println( x);   // wrong value with the macro

  x = 15.0;
  x = round( x) / 10.0; // always calculated with floats
  Serial.println( x);   // always okay

  x = 15.0;
  x = round( x);
  x /= 10;              // always calculated with floats
  Serial.println( x);   // always okay

  x = 1.5E+20;
  x = round( x);        // the macro fails
  x /= 1E+20;           // make it printable with Serial.println
  Serial.println( x);   // wrong value with macro 
}

void loop()
{
}
```

matthijskooijman, 2020-10-01:
Is this still an issue? It seems that currently, no `round` macro is defined at all anymore and `math.h` is included.

Not sure if this was really intentional, since the commits are big and the commit messages do not mention these changes at all. See e7cda3c9c906a22bcd0ab5906942546649660430 where `math.h` was added and 47e23ccfe5062e6075eb6d5a11be49d909f99ea3 where `round` was removed.

Koepel, 2020-10-01:
That is the ArduinoAPI.h
When I look into the current version of Arduino IDE 1.8.13, then the macro for abs() and round() are still in Arduino.h.

ThFischer, 2021-02-16:
IMHO the round() macro is a nasty pitfall.
It took me a couple of hours of debugging my code to figure out that it produces wrong results like
```
float a = round(10.0f);  
Serial.println(a);   // ==> 10.50
```
I propose to fix this.

Koepel, 2021-02-17:
@ThFischer Please give a working sketch and tell us which board and which Arduino IDE version or build environment you use. I can not verify that with a Arduino Uno and with Arduino IDE 1.8.13.

ThFischer, 2021-02-17:
@Koepel 
Hi, my environment is
- Visual Studio Code v1.53.2 (macOS)
- PlatformIO v2.2.1
   platformio.ini:
    ```
    platform = atmelavr
    framework = arduino
    board = ATmega328P
    ```

IntelliSense points me to the `round()` definition in MiniCore v2.0.9 (.platformio/packages/framework-arduino-avr-minicore/cores/MiniCore/Arduino.h)

PaulStoffregen, 2021-02-18:
I could not reproduce the problem on MKR1000, Arduino Due, or Arduino Leonardo using Arduino 1.8.13.

I tested using this:

```
void setup() {
  Serial.begin(9600);
  while (!Serial) ; // wait
  Serial.println("round test");
  float a = round(10.0f);  
  Serial.println(a);   // ==> 10.50
}

void loop() {
}
```

Maybe the bug is with MiniCore or PlatformIO rather than Arduino?

matthijskooijman, 2021-02-18:
@ThFischer, what you're seeing is a completely different problem, and since you're not actually using an official Arduino core and not a core based on this repository, I think this is not the right issue or repo to report it. More specifically, and assuming you're using the MCUDude minicore, it seems there is a a bug in their round macro (missing parenthesis, they convert to long *before* adding 0.5, which prevents it from working): https://github.com/MCUdude/MiniCore/blob/51ed51cd6e00529171e39dbee95d657ad2ef621b/avr/cores/MCUdude_corefiles/Arduino.h#L128

    #define round(x)     ({ typeof (x) _x = (x); _x >= 0 ? (long)_x + 0.5 : (long)_x - 0.5; })

@ThFischer I would suggest reporting this as a bug in that repository instead, so we can keep this issue focused on the actual round macro in Arduino itself, and whether it should return integer or float.

matthijskooijman, 2021-02-18:
Back to the original subject, @Koepel said:

> That is the ArduinoAPI.h
> When I look into the current version of Arduino IDE 1.8.13, then the macro for abs() and round() are still in Arduino.h.

I suspect you are looking at the AVR core? That is not yet based on this ArduinoCore-API repository, but as soon as it is converted, it should use code from this repo and the macro is, IIUC, gone in favor of using `math.h`. To verify this, you can already use the megaavr and samd cores, which are already based on this repo.

My original question about whether this is intentional or not still stands, though, since this change does have small compatibility implications, so it should be at least documented somewhere.

per1234, 2026-04-26:
Closing as resolved by https://github.com/arduino/ArduinoCore-API/commit/e7cda3c9c906a22bcd0ab5906942546649660430 and https://github.com/arduino/ArduinoCore-API/commit/47e23ccfe5062e6075eb6d5a11be49d909f99ea3 (thanks to Matthijs for pointing that out!).

It is true that the problem remains in the "Arduino AVR Boards" core, but tracking that is off topic for this repository's issue tracker. An issue in the [`arduino/ArduinoCore-avr` repository](https://github.com/arduino/ArduinoCore-avr) is welcome.