arduino/ArduinoCore-API #93: Should provide DMA library

matthijskooijman, 2019-09-16:
The SAMD core supports DMA, but this cannot be easily used from a library or sketch currently. In particular, multiple users of the DMA module need some coordination about what channels are in use, where the descriptor memory is located (I think), when to reset the DMA controller, etc.

Currently, I think there is the [Adafruit_ZeroDMA](https://github.com/adafruit/Adafruit_ZeroDMA) library that does this, but the [I2S library](https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/I2S/src/utility/DMA.h) also has a DMA coordinator embedded (so these two will not play together nicely).

A single official library contained in the core (or perhaps an API contained in the core itself) would be good here. This could be a fully featured DMA library, but it might also be useful to have a minimal DMA coordinator library (that e.g. only handles resetting/initializing the controller, allocating channels and descriptors and maybe more, leaving all the channel-specific configuration that can coexist to third party or other libraries).

PaulStoffregen, 2019-09-16:
Teensy has a DMA abstraction and management of which channels are allocated in its core library.  Here's the main header file.

https://github.com/PaulStoffregen/cores/blob/master/teensy3/DMAChannel.h

Teensy's approach tries to create Arduino style functions to specify what the DMA transfer does in a hardware-independent way.  The functionality is similar to Adafruit's library, but the API style is quite different.  Adafruit's library depends pretty heavily on a number of Atmel's defines in utility/dma.h.  Those Atmel constants might not be the best long-term for any API meant to work across all platforms.

matthijskooijman, 2019-09-20:
@sandeepmistry has transferred this issue from the SAMD repo to the API repo, but I am not quite sure if this is an API that can be generalized across architectures. I do not have much experience with DMA hardware, but I suspect that different architectures use significantly different DMA features and terminology, so any user of such a DMA library must already be target-specific.

Having said that: If we can find a generic API that is meaningful and can be implemented on all (DMA-supporting) architectures (possibly augmented by target-specific additions where needed), that would of course be great.