arduino/ArduinoCore-API #102: Why is flush() in Stream and not in Print?

jandrassy, 2019-11-08:
Why is in Stream.h `virtual void flush() = 0;`?
`flush()` is output related and should be in Print

matthijskooijman, 2019-11-09:
Hm, it surprises me that it is. In the AVR and SAM cores, this method was moved to `Print` a long time ago: https://github.com/arduino/Arduino/pull/2387

I wonder what version this API was based on, or if it was maybe moved back for some reason?

PaulStoffregen, 2019-11-09:
This repository also seem to *still* pre-date the addition of availableForWrite().  I mentioned this ~2 years ago.  https://github.com/arduino/ArduinoCore-API/issues/64#issuecomment-159860430  Maybe it should be its own issue?  But frankly, I've having a hard time taking this repository seriously.  :(

jandrassy, 2019-11-09:
> This repository also seem to _still_ pre-date the addition of availableForWrite(), which has been a part of the core library for a very long time. I mentioned this ~4 years ago. [#64 (comment)](https://github.com/arduino/ArduinoCore-API/issues/64#issuecomment-159860430) Maybe it should be its own issue? But frankly, I've having a hard time taking this repository seriously. :(

but it is used in megaavr core and in mbed core

bxparks, 2019-11-11:
It is even more confusing that:
* the ESP32 core defines `flush()` in its [Stream.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Stream.h#L51), while
* the ESP8266 core defines `flush()` in its [Print.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.h#L103).

I had to use `#if` conditional macros to create code that worked across AVR, ESP8266, ESP32, etc. I would have thought that something as fundamental as `Print` would have a consistent API across all the Arduino platforms.

jandrassy, 2019-11-11:
https://github.com/espressif/arduino-esp32/issues/943

jandrassy, 2019-11-11:
> It is even more confusing that:
>
>     * the ESP32 core defines `flush()` in its [Stream.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Stream.h#L51), while
>
>     * the ESP8266 [...]

https://github.com/espressif/arduino-esp32/issues/943

closed 2019-11-13 by cmaglie

jandrassy, 2019-11-13:
and in the `namespace arduino` branch for mbed core?

facchinm, 2019-11-13:
I'm rebasing it there too

PR #104 (2019-11-13): Some Print / Stream API fixes
Moved `flush` method to `Print`.
Added `availableForWrite` method to `Print`.

Fix #102