arduino/ArduinoCore-API #111: Uninitalised memory when copying String with embedded NUL character

cvwillegen, 2020-04-11:
When you create a String object with an embedded NUL character, and you copy this String, the memory after the NUL byte is not copied, leading to uninitialised memory being used.

Here's sample code to show the error:
```
String sGlobal;

void dumpString(const String &s)
{
  Serial.print("Got a string of length ");
  Serial.println(s.length());
  Serial.print(">");
  for (size_t t = 0; t < s.length(); ++t) {
    if (s.charAt(t) != '\0' && isascii(s.charAt(t))) {
      Serial.print(s.charAt(t));
    } else if (s.charAt(t) == '\0') {
      Serial.print("\\0");
    } else {
      Serial.print("\\x");
      Serial.print(s.charAt(t), HEX);
    }
  }
  Serial.print("<");

  Serial.println();
}

void encode(String &s)
{
  while (s.length() < 12)
  {
    s += ' ';
  }

  Serial.println("s in encode is, after filling with spaces:");
  dumpString(s);

  s.setCharAt(11, '!');
  s.setCharAt(10, 'd');
  s.setCharAt(9, 'l');
  s.setCharAt(8, 'r');
  s.setCharAt(7, 'o');
  s.setCharAt(6, 'w');
  s.setCharAt(5, '\0');
  s.setCharAt(4, 'o');
  s.setCharAt(3, 'l');
  s.setCharAt(2, 'l');
  s.setCharAt(1, 'e');
  s.setCharAt(0, 'H');
  
  Serial.println("s in encode is, after setting its chars:");
  dumpString(s);

}

void test() {
  String sLocal;
  Serial.println("sLocal in test is, after init:");
  dumpString(sLocal);
  Serial.println("sGlobal in test is, after init:");
  dumpString(sGlobal);
  
  encode(sLocal);
  Serial.println("sLocal in test is, after encode:");
  dumpString(sLocal);
  Serial.println("sGlobal in test is, after encode:");
  dumpString(sGlobal);

  sGlobal = sLocal;
  Serial.println("sGlobal in test is, after assignment:");
  dumpString(sGlobal);
}

void setup()
{
  Serial.begin(115200);

  test();
}

void loop()
{
}
```

Output of the code:
```
Local in test is, after init:
Got a string of length 0
><
sGlobal in test is, after init:
Got a string of length 0
><
s in encode is, after filling with spaces:
Got a string of length 12
>            <
s in encode is, after setting its chars:
Got a string of length 12
>Hello\0world!<
sLocal in test is, after encode:
Got a string of length 12
>Hello\0world!<
sGlobal in test is, after encode:
Got a string of length 0
><
sGlobal in test is, after assignment:
Got a string of length 12
>Hello\0n\xFFFFFFEF\xFFFFFFD6\xFFFFFFFF\<
```

Please merge #97 to fix this issue, or at least use ``memcpy()`` instead of ``strcpy()`` to initialise the data.

sebromero, 2021-03-11:
@facchinm #97 was merged, this can be closed I guess.

matthijskooijman, 2021-03-11:
I guess so, but it would be good to doublecheck. @cvwillegen Could you maybe recheck with the latest master to see if your testcase now works as expected?