arduino/ArduinoCore-API #112: String == String does not work with embedded NUL characters

cvwillegen, 2020-04-18:
https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L445 and https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L445 should use ``memcmp()`` to make sure that a String object can be compared to another String object that has embedded ``NUL`` characters

cvwillegen, 2020-04-18:
Also, https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L467 https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L492 https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L498 have the same problem.

You can debate if there is a change needed to https://github.com/arduino/ArduinoCore-API/blob/7f8de5869ee3de7a915bf1db64313c08595b3ac3/api/String.cpp#L470, but ``String::equalsIgnoreCase`` implies only printable characters.

matthijskooijman, 2020-04-21:
Did you see #97? It is related and I think it might solve your issue (but it's been too long since I wrote it, so I cannot recall exactly).

cvwillegen, 2020-04-21:
Yes, I saw your pull request (and commented on it). As far as I can see, it
still has strcmp() calls in it... please correct me if I'm wrong!

Christ van Willegen

Prath06, 2020-11-03:
Instead of strcmp(), try using stricmp(). It may work

matthijskooijman, 2020-11-04:
@cvwillegen, seems you're right, I should probaly dust off that PR sometime and also fix those calls...

@Prath06, but it seems that `stricmp()` just makes the comparison case insensitive, and also it does not seem to be a standard libc function at all. How would this help here?

mrengineer7777, 2022-05-04:
In C, C strings are just an array of chars terminated by NULL.  That is how strcmp(), strcpy(), etc know where the end of the string is.  The Arduino String class does a nice job hiding those implementation details, but the fact remains: embedded nulls in Strings WILL BREAK THINGS.