arduino/ArduinoCore-API #114: File base class

PaulStoffregen, 2020-05-17:
Similar to Print, Stream, Client, Server... we probably need a File base class.

Currently File is defined in the SD library, and also in the Bridge library for Yun.  Most programs in the Arduino ecosystem which use files are tied to the SD library.  Because they include SD.h in their headers, the SD library is automatically brought into the project.

But the SD library isn't the only possible source for files.  Flash chips, USB mass storage, network filesystems can also provide files.  Without a base class, each library using files needs to be edited.

Use of alternate SD card libraries, like SdFat, is also very difficult without a base class because so many libraries which consume files are SD.h included by their code.

Hopefully the time has come to move File out of the SD library and into an abstract base class?

PaulStoffregen, 2020-10-31:
I've recently been working on File base class for Teensy.  Details here:

https://forum.pjrc.com/threads/64136-File-abstraction-and-SdFat-integration

Turns out the ESP8266 and ESP32 cores have implemented a File base class for quite some time.  Looks like they ran into similar "C++ slicing" issues and solved them with a complicated set of classes which seem to build on top of a virtual filesystem layer inside their SDK.

I also looked the the Portenta core today.  The Portenta_SDCARD library has a TestSDCARD example which looks pretty much nothing like Arduino's API.  :(

fabiuz7, 2021-04-08:
Any news on File and File System? Also referenced here #16 
I have looked at esp8266 and esp32, they seem pretty clear and intuitive. In ESP8266 core, there is also the concept of directory (`Dir` class).