arduino/ArduinoCore-API #116: shiftIn return value misleading

oclyke, 2020-07-16:
According to the [shiftIn documentation](https://www.arduino.cc/reference/en/language/functions/advanced-io/shiftin/) the return parameter should be ```byte```. For symmetry with the ```shiftOut``` function it would make sense to return ```uint8_t```. 

In this API the return value is currently specified as ```pin_size_t``` which can be either ```uint8_t``` or ```uint32_t```

``` c++
#ifdef EXTENDED_PIN_MODE
// Platforms who wnat to declare more than 256 pins need to define EXTENDED_PIN_MODE globally
typedef uint32_t pin_size_t;
#else
typedef uint8_t pin_size_t;
#endif
```

Suggest changing the declaration from:
``` c++
pin_size_t shiftIn(pin_size_t dataPin, pin_size_t clockPin, BitOrder bitOrder);
```

to:
``` c++
uint8_t shiftIn(pin_size_t dataPin, pin_size_t clockPin, BitOrder bitOrder);
```

Any reason not to do this?

facchinm, 2020-07-22:
@oclyke you are TOTALLY right. Can you open a PR for proper attribution? Thanks!

oclyke, 2020-07-22:
You bet, thanks @facchinm! (btw all the new ```ArduinoCore-``` repos are a nice system... keep up the good work!)

closed 2020-07-22 by facchinm