arduino/ArduinoCore-API #124: Porting guide

beriberikix, 2020-09-26:
Is there an existing or planned guide on how to port to platforms? I'm specifically interested in new RTOSes (see discussion on zephyrproject-rtos/zephyr#22247) but I imagine the same guide would be helpful for anyone trying to bring up a new baremetal platform using the Arduino core API.

alranel, 2021-12-08:
Hi @beriberikix - sorry for the late answer. There's no guide at the moment, but it would be very nice to have one. I think a good starting point it the github.com/arduino/ArduinoCore-mbed repository which shows how a platform implementing this API can be developed.
Maybe could you help drafting such guide? :)

beriberikix, 2021-12-09:
No worries @alranel! I've also found the `arduin-cli` to have little nuggets of good information, especially the [Platform specification](https://arduino.github.io/arduino-cli/0.20/platform-specification/) section.

Your reply is actually well time - my team and are currently exploring this space! Would you be open to chatting further?

beriberikix, 2022-03-15:
Hello! As a small update I submitted a related project for the Google Summer of Code (GSoC) under the linux foundation: https://wiki.linuxfoundation.org/gsoc/2022-gsoc-zephyr#arduino_module_based_on_zephyr

beriberikix, 2022-12-17:
@alranel bringing this up again as we'd really love help with the IDE/CLI integration. A porting guide is a lot of work so perhaps someone can develop a "minimal" Core implementation that we can study?

aentinger, 2022-12-28:
>Hello! The GSoC season has ended and we've achieved a lot. The Zephyr Core can be found at [zephyrproject-rtos/gsoc-2022-arduino-core](https://github.com/zephyrproject-rtos/gsoc-2022-arduino-core). Key Arduino APIs are implemented as well as the ability to call into the Zephyr APIs for advanced usage.

Well, this is pretty cool :sunglasses: :+1: 

>@alranel bringing this up again as we'd really love help with the IDE/CLI integration. A porting guide is a lot of work so perhaps someone can develop a "minimal" Core implementation that we can study?

I'd suggest analyzing a simple 3rd party core, i.e. [arduino-pico](https://github.com/earlephilhower/arduino-pico) which is a 3rd party core targeting only a single architecture (and MCU for that matter), the RP2040.