arduino/ArduinoCore-API #136: Client.h not included from AruinoAPI.h

bucienator, 2020-12-22:
Some libraries, like WiFi101 are not ready to work together with the new API architecture. They include headers directly, like "IPAddress.h", "Print.h" or "Client.h" This works with Arduino IDE, as compiling there puts the "cores/arduino/api/deprecated/" directory onto the include path. However, using PlatformIO does not include the "deprecated" directory, so I had to do some changes to the libraries to work with new style frameworks. (This is fine, or at least an issue with the libraries, not the ArduinoCore-API)

The issue related to ArduinoCore-API is that it seems the recommended way for including the api is to include "Arduino.h", which would include "ArduinoAPI.h", which would then include all the other APIs. This works for most APIs, however, not for Client, as "Client.h" seems to be not included from "ArduinoAPI.h".

I didn't create a pull request, as I'm not sure if this really is a bug, or I was mislead somehow, and I should have included Client.h in a different way. Can you please provide some insight on this?

Thanks!

facchinm, 2020-12-23:
Hi @bucienator ,
thanks for reporting! We are still polishing the whole API migration so every feedback is important.
Client.h inclusion was removed with [this](https://github.com/arduino/ArduinoCore-API/commit/36b6f53a9b0eb287f4e7911c730d99f53ee66821) commit since it clashed with some libraries redefining Client class (can't find the exact issue right now but will search a bit and cross reference it).
Beside that, platform.io should respect the rules in `platform.txt` AFAIK, so it's a bit awkward that it doesn't include the content of `deprecated` folder... 
Probably filing them a bugreport is the way to go :slightly_smiling_face:

bucienator, 2020-12-25:
Originally I had the issue on Mac, so I tried to replicate it on Windows before reporting to platform.io, but then all worked well on Windows. So I went back to Mac, and retried it once with a clean install (removing ~/.platformio and <projectdir>/.pio directories, and then it worked on the Mac too. Even though I recall trying the clean run before reporting this issue. Anyway, it seems my case was an artifact of an update, and a clean run solved the issue.

Thanks for your reply!

closed 2020-12-25 by bucienator

bblanchon, 2023-03-04:
Hi guys,

Sorry for resurrecting such an old issue but `Client.h` is not included in `ArduinoAPI.h`.
It used to be but was removed in 36b6f53, as part of #120.

Can we put it back in?

Best regards,
Benoit

aentinger, 2023-03-06:
Hi @facchinm do you have any idea via @giulcioffi removed `Client.h`? It's interesting because removing `Client.h` does not really fit the scope of the #120.

On the other hand, its been removed for quite some time. @bblanchon what exact use case is this breaking for you? You simply express the desire to undo an old change, without providing any background on why and what's your special use case?

bblanchon, 2023-03-06:
@aentinger, as I said in #120, [deprecated/Client.h](https://github.com/arduino/ArduinoCore-API/blob/master/api/deprecated/Client.h#L19) contains the following comment:

> including Client.h is deprecated, for all future projects use Arduino.h instead

As I was trying to do that in the [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) library, I discovered that it doesn't work for `Client.h`.

So it's not breaking any use case for me, thanks for asking; it just contradicts your own injunctions.

aentinger, 2023-03-20:
I'd be allright with re-adding it, @facchinm any objections?