arduino/ArduinoCore-API #139: Add the ability to assign a value, or add/subtract time,  to millis() and micros()

EternityForest, 2021-02-03:
Currently there are many assorted low-power libraries, but all of them generally interfere with millis() and micros().

If an API were provided to add or subtract time to these counters, then anyone implementing deep sleep could simply add the approximate sleep duration, keeping millis() dependent applications functional, at least where precision is not needed.

In addition, more complex things like getting the sleep time from an RTC are possible, so the approach still works in precision scenarios.

Going further, applications in need of precise timing could use GPS to correct the millis() and micros count, and timekeeping libraries that use millis() to simulate an RTC could then take advantage of this.

Going even further, the ability to explicitly set the times may make certain kinds of testing and debugging easier.

The API could simply be addElapsedTime(int32_t), setMicros(uint32_t), and setMillis(uint32_t), none of which require any overhead in the interrupt handlers, or anything the optimizer can't just get rid of, if not needed.  

addElapsedTime would affect both micros() and millis(), and serve as the obvious, well-defined way to inform the system that you have just woken from deep sleep for a length of time.

matthijskooijman, 2021-02-03:
This was previously reported as #69, so I'm going ahead to close this as duplicate to keep discussion in one place. Maybe you could add some of your specific suggestions to the other issue?

closed 2021-02-03 by matthijskooijman

EternityForest, 2021-02-03:
Will do!  Can't believe I didn't see that one while I was searching before
I posted!

matthijskooijman, 2021-02-03:
> Will do!  Can't believe I didn't see that one while I was searching before I posted!

Well, we did sneakily move your issue from the primary arduino/Arduino repository to arduino/ArduinoCore-API repo, which is where the API is now being developed, so you were probably searching in the wrong place. No worries though! :-)