arduino/ArduinoCore-API #141: Each core provides a different USBAPI

Legion2, 2021-02-18:
The Arduino Core API defines a very low level USB api in [api/USBAPI.h](https://github.com/arduino/ArduinoCore-API/blob/master/api/USBAPI.h). This API is implemented by the avr core in [cores/arduino/USBCore.cpp](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/USBCore.cpp). But it is not implemented by samd core, which implements the same functionality but in its own [`USBDeviceClass` class](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/USB/USBCore.cpp). The USB implementation in megaavr core (`USBCore.cpp`) is a copy of the avr usb implementation with some deviations. The mbed core implements its own version of the `PluggableUSBModule`.

As a result, each core implements its own USB API which is incompatible with the others, there exist many code duplicates and users of the USB API have to understand and adapt their code to each individual core implementation.

Are there any plans to improve this situation?

dogtopus, 2022-04-07:
This is one of the biggest reason if not the only reason I don't like Arduino despite its quite mature ecosystem.

per1234, 2026-09-28:
Unless I miss something, this is not actionable from the codebase hosted in this repository. The necessary work has been done here of defining a canonical standard API. The remaining work is in each of the individual platform codebases.

You are welcome to open dedicated issues for that in the relevant platform repositories.

closed 2026-09-28 by per1234