arduino/ArduinoCore-API #142: Concat operator + with random() value leads to crash

sebromero, 2021-03-11:
I found that when I concatenate a string directly (without wrapping it in a String object) with a number obtained from `random()` I get strange results.
When `random()` is used with a parameter, the resulting string contains garbage obtained from memory where it probably shouldn't read from.
When using `random()` without parameter the program crashes. On Portenta I get the red S.O.S flash, on the MKR1000 for example it just stops responding. I started to investigate but haven't found the culprit yet. It crashes somewhere in `concat()` when reading the string length using `strlen()`.

```cpp
void setup() {  
  Serial.begin(9600);
  while(!Serial);  

  //String newMessage = "Hello World " + random(1024); // Prints garbage:3mc⸮Envie M7
  String newMessage = "Hello World " + random(); // Crashes
  //String newMessage = "Hello World " + String(random()); // Works
  //String newMessage = "Hello World " + String(random(1024)); // Works
  
  Serial.println(newMessage);
}
  
void loop() {}
```

matthijskooijman, 2021-03-11:
Seems you reported this twice, closing this one as duplicate of #143.

closed 2021-03-11 by matthijskooijman

sebromero, 2021-03-12:
Oh wow, one of the rare cases of messing up the state machine :-) Connection stalled while the browser was sending the request so I switched the WiFi which caused double posting. I'm glad it didn't happen during an e-banking transfer 😅