arduino/ArduinoCore-API #155: Extending PinMode to support MCUs with open-drain outputs?

obra, 2021-09-15:
I'm in the process of collaborating with some other folks to build a new Arduino port for GigaDevice's GD32 chips[1].  We're attempting to build on top of ArduinoCore-API. In general, It's so much better than the old way of doing things. One thing we're running into is how to handle gaps in the PinMode enum.

Specifically, we're short an `OUTPUT_OPENDRAIN` in `PinMode` 

Would a PR that adds a new `OUPUT_OPENDRAIN = 0x4` to the PinMode definition be welcome? If not, is there an alternate "Arduinoish" way to build the user-facing API to set a pin as an open-drain output?

[1] https://github.com/CommunityGD32Cores/ArduinoCore-GD32

Thanks!

facchinm, 2021-09-22:
I'm totally :+1: to add the `OUTPUT_OPENDRAIN` define; would you mind to file a PR with the change?

obra, 2021-09-22:
Done as #157, with copious notes for the next person who needs to work around lack of some pin mode in the commit message ;)

Based on the implementation of Analog pins in the Arduino world, I'm guessing that a standard "INPUT_ANALOG" PinMode is *not* of general interest?

PR #157 (2021-10-01): Add a new 'OUTPUT_OPENDRAIN' pinmode for platforms.
Not every MCU offers open drain pins, but some do. This change
means that they'll no longer have to play games to implement an
API-compatible pinMode() function that offers open drain.

Fixes #155

If you're reading this commit message because you need to add another
pin mode to this enum, the hacky trivial workaround is to do something
like this in your core:

```
// This typedef is used to extend the PinMode typedef enum
// in the ArduinoAPI, since they don't have contants
typedef enum {
        INPUT_ANALOG = 99 , // We assume that the Ard [... cut]
commit df98331b (2021-10-01): Add a new 'OUTPUT_OPENDRAIN' pinmode for platforms.

Not every MCU offers open drain pins, but some do. This change
means that they'll no longer have to play games to implement an
API-compatible pinMode() function that offers open drain.

Fixes #155

If you're reading this commit message because you need to add another
pin mode to this enum, the hacky trivial workaround is to do something
like this in your core:

```
// This typedef is used to extend the PinMode typedef enum
// in the ArduinoAPI, since they don't have contants
typedef enum {
        INPUT_ANALOG = 99 , // We assume that the Ar [... cut]
closed 2021-10-01 by facchinm