arduino/ArduinoCore-API #178: Serial.print(0ULL) gives not output

warmonkey, 2023-01-14:
Test case:
`Serial.print(0ULL);`
Expected output: 0
Actual output: nothing

Quick fix:
```
// FAST IMPLEMENTATION FOR ULL
size_t Print::printULLNumber(unsigned long long n64, uint8_t base)
{
  //handle special case
  if (n64 == 0) {
    write('0');
    return 1;
  }
```

aentinger, 2023-01-16:
Can you create a PR or at least a failing test case?

earlephilhower, 2023-09-03:
Failing example from https://github.com/earlephilhower/arduino-pico/issues/1691

````
void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  uint64_t n = 123;
  uint64_t m = n;
  Serial.print("n - m = ");
  Serial.println(n - m);
  n++;
  Serial.print("n - m = ");
  Serial.println(n - m);
}
void loop() {}
````

Gives
````
n - m =
n - m = 1
````

PR #203 (2023-09-04): Fix of empty string printed in case of a 64-bit unsigned integer.
This fixes #178 .
closed 2023-09-04 by aentinger