arduino/ArduinoCore-API #241: Bug in I2C slave STM32F103

AndKe, 2024-11-30:
### Description

```
#include <Wire.h>
long lastms;
volatile int x = 0;
volatile int y = 0;

#define STM
void receiveEvent(int howMany) {
  x = Wire.read();
}

void requestEvent() {
  y = 1;
}

void setup() {
  Serial.begin(115200);

  #ifdef STM
  //STM32F103
  pinMode(PB6, INPUT); //PB6=SCL
  pinMode(PB7, INPUT); //PB7=SDA
  Wire.setSCL(PB6);
  Wire.setSDA(PB7);
 
  #elif 
  //Arduino Nano
  pinMode(A4, INPUT);
  pinMode(A5, INPUT);
  //Wire.setSDA(A4);
  //Wire.setSCL(A5);

 #endif
  
 
  Wire.begin(11);                // join i2c bus as slave
  //Wire.onRequest(requestEvent);  
  Wire.onReceive(receiveEvent); 

  lastms = millis();
}

void loop() {
  if (millis() > lastms + 1000) {
    Serial.print("-");
    lastms = millis();
  }
  if (x != 0) {
    Serial.print("U:0x");
    Serial.println(x, HEX);
    x = 0;
  }
  if (y != 0) {
    Serial.println("V");
    y = 0;
  }
}
```

### Environment

- Boards platform name: Bluepill / own board  / Arduino Nano (Nano and Leonardo works fine)
- Boards platform version (as shown in Boards Manager):  [image]

- ArduinoCore-API version (if you manually installed it):  

### Current behavior

If compiled for Nano, the application works. (displaying the incoming I2C requests.)
When compiled for STM32F103 - with CDC serial, it does not. (it outputs only the "-" each 1sec.)

The incoming data is at the same time decoded by an oscilloscope, so pullups etc are ok.

### Expected behavior

I expect the code to work on STM32F103C8 /Bluepill  like it did on Nano

### Additional information

When on Bluepill, On incoming I2C data, PB6 (SCK) is being pulled low, and held low by the controller, effectively killing all further traffic.
This happens even if USB /Serial support is set to NONE

per1234, 2024-11-30:
Hi @AndKe. This repository only defines the standardized core `HardwareI2C` class API:

https://github.com/arduino/ArduinoCore-API/blob/master/api/HardwareI2C.h

Since it is very architecture-specific, the class and the "**Wire**" library can not be implemented in the universal code hosted in this repository. It is implemented independently in each Arduino boards platform.

So I will close this as off topic. You are welcome to request assistance with the problem over on Arduino Forum:

https://forum.arduino.cc/c/using-arduino/networking-protocols-and-devices/27

closed 2024-11-30 by per1234

AndKe, 2024-12-01:
After several days of chasing /verifying  this nasty bug I did, ask in the forum: https://forum.arduino.cc/t/bug-in-i2c-slave-stm32f103-please-verify/1327398

there is very little help to get. 

I do understand that you cannot/will not cover every chip  MCU, but it would be very nice of you to throw a compiler warning if Wire Slave is used with for exampled STM32F103 series - so that less time was wasted on things like this.

JAndrassy, 2024-12-01:
https://github.com/stm32duino